STMicroelectronics/stm32ai-modelzoo #2: Conflicting requirements.txt

mridup, 2023-02-09:
The requirements.txt has conflicting package version numbers once it is iinstalled.
E.g.
ERROR: numba 0.56.4 has requirement numpy<1.24,>=1.18, but you'll have numpy 1.24.2 which is incompatible.
ERROR: onnx 1.13.0 has requirement protobuf<4,>=3.20.2, but you'll have protobuf 3.19.6 which is incompatible.
ERROR: skl2onnx 1.13 has requirement scikit-learn<=1.1.1, but you'll have scikit-learn 1.2.1 which is incompatible.

If we correct the above versions, then more conflicting versions emerge.

Do you have a fixed or frozen requirements.txt file which works for the training phase as required in the HAR example?

Shahnawax, 2023-02-10:
We did not manage to reproduce the reported issue, (tested with Python versions 3.9 and 3.10). Could you (re)try to create a clean Python or Conda virtual environment and install the requirements in it using the following command? Try using Python versions 3.9 or 3.10.
`pip install -r requirements.txt --upgrade-strategy only-if-needed`
If the issue is still not resolved, kindly share the list of the installed packages along there versions?
You can get it using `pip list` command (launched in your virtual environment).

mridup, 2023-02-10:
@Shahnawax thanks for checking this out.
Actually I used python3.8 and I did use a completely new/clean virtualenv for this purpose.
I can surely try again using python3.9/3.10.
And I will report back the result. Thanks!

Shahnawax, 2023-02-16:
@mridup  could you confirm if the issue is resolved when you use the Python v3.9/3.10?

Shahnawax, 2023-02-23:
The issue is resolved by using the Python version 3.9 or 3.10.

closed 2023-02-23 by Shahnawax