STMicroelectronics/stm32ai-modelzoo #3: Cannot login to stm32ai cloud.

Alberto-SM, 2023-02-10:
After successfully login several times, now the `login` function of the LoginService gets stuck here:
`        resp = s.get(
            url=provider + "/as/authorization.oauth2",
            params={
                "response_type": "code",
                "client_id": client_id,
                "scope": "openid",
                "redirect_uri": redirect_uri,
                "response_mode": "query"
            },
            allow_redirects=True,
        )
`

hamitiya-STM, 2023-02-14:
Hello @Alberto-SM,
I am not able to reproduce this issue on my side.
First, are you able to login through [STM32Cube.AI Developer Cloud](https://stm32ai-cs.st.com) ?
If yes, can you please ensure you've correctly set the following, if you are working behind a corporate proxy:
(ex: in a shell terminal)
```sh
export HTTP_PROXY="your-proxy" # example: http://username:password@addr:port if it requires authentication
export HTTPS_PROXY="your-proxy" # example: http://username:password@addr:port if it requires authentication
export http_proxy="your-proxy" # example: http://username:password@addr:port if it requires authentication
export https_proxy="your-proxy" # example: http://username:password@addr:port if it requires authentication
export NO_SSL_VERIFY="1"
```

To be sure if your internet configuration is correct, you can verify if `curl` requests respond with a Status 200:
```sh
curl -I https://stm32ai-cs.st.com
```

If not, can you please share a screenshot/log with the error you encounter (without sensitive informtions)

Best regards,
Yanis

Alberto-SM, 2023-02-17:
Hello,
Thanks for your help. 

I can login through the web application.
The `curl` command returns 200 OK.

I don't have a specific error or log screen. I am using the tools in _common/stm32ai_. My code is practically one line:

`ai = Stm32Ai(CloudBackend(username, password))`

This call never returns as the `login` method of the `LoginService` does not have a timeout. It hangs here:

`        # Get connection initialization procedure
        resp = s.get(
            url=provider + "/as/authorization.oauth2",
            params={
                "response_type": "code",
                "client_id": client_id,
                "scope": "openid",
                "redirect_uri": redirect_uri,
                "response_mode": "query"
            },
            allow_redirects=True,
        )`

I'll keep investigating what is going on.

hamitiya-STM, 2023-02-23:
Hello Alberto,
We pushed a newer version of the model zoo which fixes an issue related to login.
Can you please retry with the latest version ?

Best regards,
Yanis

Alberto-SM, 2023-02-26:
The fix solved the issue.
Thanks a lot!

closed 2023-02-26 by Alberto-SM