STMicroelectronics/stm32ai-modelzoo #4: Stuck after entering password

TruongHaiDang, 2023-02-22:
I am training object detection with stm32ai-model zoo, I use the train scripts with these configs in user_config.yaml. I have an STM account and this account can log in to the stm32ai cloud, the path to stm32ai.exe is correct too.
After entering the password I stuck in this screen. Help me to solve it. Thanks a lot.

This is my user_config.yaml:
```
general:
  project_name: FireProtection
  logs_dir: D:/GitHub/FireProtection/training/logs
  saved_models_dir: D:/GitHub/FireProtection/training/output

train_parameters:
  batch_size: 64
  training_epochs: 10000
  optimizer: adam
  initial_learning: 0.001
  learning_rate_scheduler: reducelronplateau

dataset:
  name: Fire
  class_names: [fire]
  training_path: D:/GitHub/FireProtection/training/dataset/train
  validation_path: D:/GitHub/FireProtection/training/dataset/valid
  test_path: D:/GitHub/FireProtection/training/dataset/test

pre_processing:
  rescaling: {scale : 127.5, offset : -1}
  resizing: bilinear
  aspect_ratio: False
  color_mode: rgb

post_processing:
  confidence_thresh: 0.01
  NMS_thresh: 0.5
  IoU_eval_thresh: 0.4

data_augmentation:
  augment: True
  rotation: 30
  shearing: 15
  translation: 0.1
  vertical_flip: 0.5
  horizantal_flip: 0.2
  gaussian_blur: 3.0
  linear_contrast: [0.75, 1.5]

model:
  model_type: {name : mobilenet, version : v1, alpha : 0.25} 
  input_shape: [256, 256, 3]
  transfer_learning : True

quantization:
  quantize: True
  evaluate: True
  quantizer: TFlite_converter
  quantization_type: PTQ
  quantization_input_type: uint8
  quantization_output_type: float
  export_dir: quantized_models

stm32ai:
  optimization: balanced
  footprints_on_target: STM32H747I-DISCO
  path_to_stm32ai: C:/Users/haida/STM32Cube/Repository/Packs/STMicroelectronics/X-CUBE-AI/7.3.0/Utilities/windows/stm32ai.exe
  
mlflow:
  uri: ./mlruns

hydra:
  run:
    dir: outputs/${now:%Y_%m_%d_%H_%M_%S}
```

This is my issue:
[image]

hamitiya-STM, 2023-02-23:
Hello Truong,
We pushed a fix related to login this afternoon. 
Can you please pull the latest version of our Model Zoo ?
Please let us informed if it fixes your issue or not.

Best regards,
Yanis

TruongHaiDang, 2023-02-24:
> Hello Truong, We pushed a fix related to login this afternoon. Can you please pull the latest version of our Model Zoo ? Please let us informed if it fixes your issue or not.
> 
> Best regards, Yanis

Thanks a lot, it worked fine on my machine. But it cannot authenticate in google colab, then it call stm32ai.exe in local enviroment in throw this error:

[image]

Will you fix it?

hamitiya-STM, 2023-02-24:
Hello, Truong,
```python
# Set environment variables with your credentials to access STM32Cube.AI Developer Cloud Services
os.environ['TruongHaiDang'] = '*******@gmail.com'
```
is wrong. Environment variable is "stmai_username".
Please try again by setting the following instead
```python
os.environ['stmai_username'] = 'something@example.com'
```

Best regards
Yanis

TruongHaiDang, 2023-02-24:
> Hello, Truong, this line
> 
> ```python
> # Set environment variables with your credentials to access STM32Cube.AI Developer Cloud Services
> os.environ['TruongHaiDang'] = '*******@gmail.com'
> ```
> 
> is wrong. Environment variable is "stmai_username". Please try again by setting the following instead
> 
> ```python
> os.environ['stmai_username'] = 'something@example.com'
> ```
> 
> Best regards Yanis

Thank you so much, this answer made it work. I will close this issue.

closed 2023-02-24 by TruongHaiDang