STMicroelectronics/stm32ai-modelzoo #5: Issue during training a model: "OSError: Unable to create file (file signature not found)."

davidroid, 2023-03-14:
Hello all, I tried to run the training of an image classification model available in the stm32ai-modelzoo, but hit the following issue: "OSError: Unable to create file (file signature not found)."

- Setup:
  - VMware Linux virtual machine: Ubuntu 22.04LTS
  - Python virtual environment 1: Python 3.10.6
  - Python virtual environment 2: Python 3.9.3
  - stm32ai service online 

- Training:
  - Guide: https://github.com/STMicroelectronics/stm32ai-modelzoo/blob/main/image_classification/scripts/training/README.md
  - Python virtual environments created with the following dependencies: https://github.com/STMicroelectronics/stm32ai-modelzoo/blob/main/requirements.txt
  - Model: MobileNetv1, 0.25, 128x128x3
  - Dataset: https://storage.googleapis.com/download.tensorflow.org/example_images/flower_photos.tgz
  - Script: https://github.com/STMicroelectronics/stm32ai-modelzoo/blob/main/image_classification/scripts/training/train.py

- Output:
  - The script runs, it configure the experiment, connect to the stm32ai service online to convert and analyze the model, then hits the following error during the training of the second epoch: "OSError: Unable to create file (file signature not found)”.

- Attachments: 
  - [user_config.yaml.txt](https://github.com/STMicroelectronics/stm32ai-modelzoo/files/10967170/user_config.yaml.txt)
  - [train_error.log](https://github.com/STMicroelectronics/stm32ai-modelzoo/files/10967169/train_error.log)

Shahnawax, 2023-03-20:
Hello @davidroid , We were not able to reproduce the issue on our side, however, doing a bit of research it looks like the issue is caused because during training we are using a set of callbacks. One of these callbacks is making sure that at every epoch it saves/updates a model checkpoint with the best validation accuracies. It looks like this file is locked for some reason. I found a similar problem [here ](https://stackoverflow.com/questions/49239961/why-is-h5py-h5f-create-failing-with-oserror-unable-to-open-file-name-name) along with the fix. Could you please try to `export HDF5_USE_FILE_LOCKING=FALSE` add this line in your `train.py` and see if it fixes the problem? The details of the solution or what it will do can be found [here](https://support.nesi.org.nz/hc/en-gb/articles/360000902955-NetCDF-HDF5-file-locking). In the meanwhile, could you please also tell us what are the versions of your OS, WSL, Python that you are using in order to try to reproduce the issues that you have? 
Let us know if the solution works, Thank you!

davidroid, 2023-03-20:
Hello @Shahnawax, I have run the training again after exporting the variable you suggested, but nothing have changed unfortunately, I got the same error.
I have updated the previous comment with the release of the OS, which is Microsoft Windows 10 Enterprise.

Shahnawax, 2023-04-05:
The issue is more likely caused due to some right issues of the file checkpoint. Discussed the fix to the reporter, and tested on multiple platforms to confirm that the issue does not exist.

closed 2023-04-05 by Shahnawax