STMicroelectronics/stm32ai-modelzoo #6: Question about machine learning

DanielMartensson, 2023-04-28:
Hi!

I have a question about your reprository about machine learning.
I'm writing all my machine learning code in pure ANSI C (C89) code and right now I'm planning to write a code base for support vector machine using quadratic programming.

https://github.com/DanielMartensson/CControl

My questions are:

1. If you would consider Support Vector Machine or Deep Neural Networks for embedded system?
2. What algorithms you are using for detection. Is it Viola-Jones algorithm?

Shahnawax, 2023-05-09:
@DanielMartensson the answers to your questions are:
1) In our repository we provide different examples for using both Support Vector Machine based Classifiers (SVC) and Deep Neural Networks. With STM32CubeAI one can deploy not only deep neural network models (h5, tflite, onnx) but also the traditional machine learning models generated using scikit-learn and exported as onnx (.onnx). To know more you can have a look on the [Human Activity Recognition](https://github.com/STMicroelectronics/stm32ai-modelzoo/tree/main/human_activity_recognition) sub-directory which provides a simple example using SVC.
2) We are not using Viola-Jones algorithm but SSD and a custom YOLO model prepared in-house by STMicroelectronics. 

I hope this answers your both questions. 

Thank you.

Shahnawax, 2023-05-09:
Answered the question.

Shahnawax, 2023-05-09:
Solved

closed 2023-05-09 by Shahnawax