STMicroelectronics/stm32ai-modelzoo #10: Title: "Error linking libneai.a and undefined reference to neai_classification in STM32CubeIDE project"

chanakyavasantha, 2023-07-06:
## Issue Description

Once we have downloaded the library zip file from Nano Edge AI Studio,Open a new stm32 project in Stm32 Cube Ide then the `libneai.a` static library file should be placed in the `Src` folder of the project. Additionally, the `NanoEdgeAi.h` and `knowledge.h` header files should be copied to the `Inc` folder. 

If we encounter an error indicating that `neai_classification`, `neai_init`, or `neai_anomaly_detection` cannot be found, it is likely that the `libneai.a` library is not accessible. To resolve this, we need to link the library with the linker ':libneai.a` and set the library search path to `../Core/Src`.

[image]

## Steps to Reproduce

1. Download the library zip file from Nano Edge AI Studio.
2. Place the `libneai.a` static library file in the `Src` folder.
3. Copy the `NanoEdgeAi.h` and `knowledge.h` header files to the `Inc` folder.
4. Build the project in STM32CubeIDE after succesfully linking libneai.a static library as shown above.

## Expected Behavior

The project should build successfully without any errors related to missing functions such as `neai_classification`, `neai_init`, or `neai_anomaly_detection`.

## Actual Behavior

Encountering errors indicating that the mentioned functions cannot be found.

## Environment

- STM32CubeIDE version: 1.12.1
- Operating System: Windows

LFOSTM, 2023-07-19:
Hello,
This issue is not related to the model zoo...
You can find NEAI GitHubs here : https://github.com/orgs/stm32-hotspot/repositories?q=nanoedge&type=all&language=&sort=
You can also use community page (or mailto:edge.ai@st.com) or using this space for more generic questions (https://stm32ai.st.com/contact/)
Regards

closed 2023-07-19 by LFOSTM

LVASTM, 2023-08-02:
Hello, 

When linking a static library, you should set only the name of the library. 
Here for libneai.a, you should write -lneai 

Regards,