STMicroelectronics/stm32ai-modelzoo #12: Why do I keep getting this error on the picture when I run Evaluate.py? How should I solve it?

znby6, 2023-07-20:
[image]

znby6, 2023-07-21:
> 你好，
> 
> 您能分享一下您使用的yaml文件吗？
[image]
[image]
[image]

Hello, this is a screenshot of my evaluate_user_config.yaml file, how can I solve this error, thank you very much.

mahdichtourou94, 2023-07-21:
Your Yaml file configuration seems good.
Can you share the quantized model + the test set in order to check if I can regenerate your issue?

znby6, 2023-07-22:
> 您的 Yaml 文件配置看起来不错。 您能否分享量化模型+测试集，以检查我是否可以重新生成您的问题？

[2023_07_20_17_47_31.zip](https://github.com/STMicroelectronics/stm32ai-modelzoo/files/12135326/2023_07_20_17_47_31.zip)
[test.zip](https://github.com/STMicroelectronics/stm32ai-modelzoo/files/12135330/test.zip)
Here is the zipped file of my quantized model and test set, any help is greatly appreciated, hope this solves my problem

mahdichtourou94, 2023-07-25:
I run the evaluation script using the provided quantized model and test set and I got the following results:

> [INFO] Evaluating the quantized model ...
[INFO] Starting calculating map
INFO: Created TensorFlow Lite XNNPACK delegate for CPU.
100%|███████████████████████████████████████████████████24/24[06:31<00:00, 16.30s/it]

> [RESULTS] AP for class crack = 0.00 %

> [RESULTS] AP for class pot = 0.00 %

> [RESULTS]` AP for class crackle = 0.00 %

> [RESULTS] AP for class Fabrics = 0.00 %

> [RESULTS] AP for class road = 0.00 %

> [RESULTS] AP for class tile = 0.00 %

> [RESULTS] Quantized model mAP = 0.00 %

> [INFO] Evaluation took 6.556 minutes

Please make sure that you are using the latest version of the project by cloning it again.

znby6, 2023-07-26:
> I run the evaluation script using the provided quantized model and test set and I got the following results:
>
>
>
>
>
>
>
>
>
>
> Please make sure that you are using the latest version of the project by [...]

Hello, I re-downloaded the latest project on github and re-configured, but the error of the previous evaluate.py still appeared. How did you run it to get the mAP calculation result? Thank you for your help！

mahdichtourou94, 2023-07-26:
Hi,
I provided the same configuration as yours in the yaml file.
Could you please precise which version of python, numpy and tensorflow you have in your environment?

znby6, 2023-07-27:
> Hi, I provided the same configuration as yours in the yaml file. Could you please precise which version of python, numpy and tensorflow you have in your environment?
Hello, I am using python version 3.8.17, numpy version 1.23.4, tensorflow-gpu 2.8.3

mahdichtourou94, 2023-07-27:
Hi,

We recommend using Python v3.9 or v3.10. as mentioned [here](https://github.com/STMicroelectronics/stm32ai-modelzoo#before-you-start)

I hope this solve your issue, let me know.

znby6, 2023-07-28:
> 你好，
> 
> 我们建议使用 Python v3.9 或 v3.10。正如[这里提到的](https://github.com/STMicroelectronics/stm32ai-modelzoo#before-you-start)
> 
> 我希望这能解决您的问题，请告诉我。

Hello, I tried to change the python version, but this error still occurred: TypeError: '<' not supported between instances of 'NoneType' and 'NoneType', May I ask how you solved this problem, I am very distressed about this problem, I have tried many methods and still can't solve it.

mahdichtourou94, 2023-07-28:
Hi,

can you please run the following commands and share with us the output:

```
pip list
```

```
python --version
```

Also which OS you are using?

mahdichtourou94, 2023-08-22:
Couldn't regenerate the issue using the same environment.
Closing the issue for now.

closed 2023-08-22 by mahdichtourou94