STMicroelectronics/stm32ai-modelzoo #13: Object Detection Demo Can Handle Multi-class?

cowbaz, 2023-08-13:
I am successfully to deploy single class. Then I have trained multi class model using training scripts provided and loaded it on board. The model final layer confirm it will output multi class result and ai config file also contain info for multi class. The mAP is about 0.46. But when running the app, it can not detect any class. Is there anything I shall config?

Many thanks.

MDUSTM, 2023-08-15:
Hello
We have indeed identified an issue with multiclass model deployment and we are investigating. We will come back to you very soon.

mahdichtourou94, 2023-08-22:
Hello,
The issue was solved with the latest updates, the deployment should work now for multi class models.