STMicroelectronics/stm32ai-modelzoo #25: which models are supported for STM32H745 board?

UpennGitttt, 2024-01-15:
I am looking for a STM32 model deployment, in the case of image classification or object detection. 
I saw that only STM32H747 is supported and I am wondering if there is any model that supports STM32H745 boards.

GRATTINSTM, 2024-01-18:
Hello @UpennGitttt, 

For object detection we are supporting only STM32H747I-DISCO boards. For image classification the NUCLEO-H743ZI2 is also supported but the card has no external memory, so it can only run the smallest models. 
We do not support STM32H745 boards. 

Guillaume

UpennGitttt, 2024-01-21:
> Hello @UpennGitttt,
> 
> For object detection we are supporting only STM32H747I-DISCO boards. For image classification the NUCLEO-H743ZI2 is also supported but the card has no external memory, so it can only run the smallest models. We do not support STM32H745 boards.
> 
> Guillaume

Thank you so much for your response! I am wondering if I can read this repo line by line and see if I can change the code so that it could train and deploy object detection or image classification models on STM32H745 board? Or there are some hidden parts that is not included in this open source repo but will be called when I run the code in this repo that can only supported STM32H747I-DISCO boards . In this case, even if I understand the all the lines of code in this repo and still not being able to train and deploy the model on the boards other than the ones specified by this repo(STM32H747I-DISCO boards). 

BTW, My goal is to train and deploy a model using customized dataset without actually entering CubeIDE.

Thank you again for your response!

UpennGitttt, 2024-01-22:
If this is possible, maybe I can investigate further on this and make a PR if I make it happens such that this could support STM32H745 board.

GRATTINSTM, 2024-01-24:
Hello @UpennGitttt, 

We will provide a new release of the ModelZoo in few weeks. The code will be deeply modified so it would be better if you waited a little before starting into these modifications. Is it the STM32H745I-DISCO board that you want to use? Which camera interface do you plan to use?

Guillaume

UpennGitttt, 2024-01-25:
> Hello @UpennGitttt,
>
> We will provide a new release of the ModelZoo in few weeks. The code will be deeply modified so it would be better if you waited a little before starting into these [...]

Thank you so much for your response, and glad to hear that a new release is coming in a few weeks! The board I plan to use is NUCLEO-H745ZI-Q. The camera I plan to use is esp32 cam. I cannot afford STM32H745I-DISCO plus B-CAMS-OMV camera, thus I want to keep the cost down as much as possible. I want to train, deploy a quantized model just like the example of STM32H745I-DISCO plus B-CAMS-OMV(the flower image classification example).  So the deployment is focused on configurations that can handle these end-to-end use cases. I could wait for a few weeks for the new release and I would like to contribute if it is possible!

Thank you!

Navodplayer1, 2024-01-25:
you can use mobilenet models for transfer learning or fine-tuning and then deploy on microcontrollers. they are lightweight and you can find pre-trained model weights for greyscale image classification.

check this repo: https://github.com/Navodplayer1/MobileNet_96x96_greyscale_weights

c

jingxan, 2024-02-01:
> Hello @UpennGitttt,
>
> We will provide a new release of the ModelZoo in few weeks. The code will be deeply modified so it would be better if you waited a little before starting into these [...]

Hello @GRATTINSTM,
May I know when the new version of ModelZoo will be released, because for now I encounter the deploy.py not working although there is no error code showing in the terminal when I run the deploy.py for image classification.

GRATTINSTM, 2024-02-23:
Hello @UpennGitttt, @jingxan, 

The new release is now available. We hope you will find it more convenient and easier to use than the previous one. 

Guillaume

ken-1025, 2024-02-23:
> 你好 ， ，
> 
> 新版本現已推出。我們希望您會發現它比以前的更方便、更易於使用。
> 
> 紀堯姆

Hi @GRATTINSTM 
I'm trying to use the new version of STM32 model zoo for object detection.
I've used the provided user_config.yaml file and made some modifications to the code.
[image]
[image]

I've successfully flashed the program onto the STM32H747I-DISCO.
However, the LCD displays as follows.
[image]

I'd like to ask you, did I miss any steps?

Ken

GRATTINSTM, 2024-02-23:
Hello @ken-1025, 

It's a misprint can you replace the class_name value by the following one: 
```yaml
  class_names: [ person ]
```
Tell me if it solves your problem. It will be quickly corrected. 

Guillaume

closed 2024-02-24 by UpennGitttt

ken-1025, 2024-02-24:
Hello @GRATTINSTM  

Yes, after modifying the class_name, the results were successfully displayed. 
However, I have encountered a new issue now. 
When deploying the model trained previously in the stm32 model zoo v1, the following error occurred.
[image]
Did I make a mistake on my end? 
I only modified the model and class_name. 
Thank you for your assistance.

Ken

GRATTINSTM, 2024-02-26:
Hello @ken-1025, 

Could you please share the model you are trying to deploy and the associated class_names ? 

Guillaume

ken-1025, 2024-02-27:
Hello @GRATTINSTM 

I think I've identified the issue. My model is using Mobilnet v1 0.25. During benchmark testing, I found that my RAM exceeded 512KB. Therefore, I attempted to reduce the number of objects.
[image]

After attempting this, I noticed that reducing the number of objects did not decrease the amount of RAM usage.
[image]
The first one has 15 objects.
The second one has 30 objects.
The third one has 36 objects.

Do you have any better suggestions for me?

Ken

GRATTINSTM, 2024-02-27:
Hello @ken-1025, 

I don't think the memory is a problem as the STM32H747I-DISCO board provide a 32-MB SDRAM and a 128-MB QSPI flash as external memories. 

Guillaume

ken-1025, 2024-02-28:
Hello @GRATTINSTM 

I really can't figure out what went wrong. 
Below are my training model and class names. 
Could you please help me take a look?
[test.zip](https://github.com/STMicroelectronics/stm32ai-modelzoo/files/14430340/test.zip)

Ken

GRATTINSTM, 2024-02-28:
Hello @ken-1025, 

I investigated your problem, the reason why the project won't compile is that the output of your model is huge ! The provided MobileNet v1 model of the ModelZoo has two classes and its output size is 30,640 KB, yours has 37 classes and its output size is 566,840 KB ! We didn't consider such a case, right now the available size for the output is fixed, it is not dynamically set as it is for the input size. 

However here is a trick to deploy your model : 
- In the _stm32ai_application_code/object_detection/Application/STM32H747I-DISCO/STM32CubeIDE/CM7/STM32H747XIHx_CM7.ld_ file, modify the sdram_section output section by adding a AppConfig_SDRAM input section inside. The sdram_section should now look like this : 
```
  .sdram_section (NOLOAD):
  {
    . = ALIGN(32);
    *(.Lcd_Display)
    *(.Lcd_Display*)
    . = ALIGN(32);
    *(.NN_Activation_Buffer_SDRAM)
    . = ALIGN(32);
    *(.CapturedImage_Buffer_SDRAM)
    . = ALIGN(32);
    *(.RescaledImage_Buffer_SDRAM)
    *(.Out_Postproc)
    *(.AppConfig_SDRAM)
  } >SDRAM
```
- In the _stm32ai_application_code/object_detection/Application/STM32H747I-DISCO/Src/CM7/main.c_ file, add an attribute to link the App_Config variable to the input section AppConfig_SDRAM : 
```
/*Application context*/
__attribute__((section(".AppConfig_SDRAM")))
AppConfig_TypeDef App_Config;
```
Thanks to these modifications, the context variable App_Config, which contains the ranking output of the model will be stored in SDRAM. 
You should now be able to deploy the model on your STM32H747I-DISCO board. Tell me if it works on your side. 

Guillaume

ken-1025, 2024-03-01:
Hello @GRATTINSTM 
Thank you for your help. 
The deployment was successful after making the modifications.

Ken