STMicroelectronics/stm32ai-modelzoo #26: Export onnx model error in stm.ai v8.1.0

aksunlight, 2024-01-29:
The error occurs when I export model [yamnet_256_64x96.h5](https://github.com/STMicroelectronics/stm32ai-modelzoo/tree/main/audio_event_detection/models/yamnet/ST_pretrainedmodel_public_dataset/esc_10/yamnet_256_64x96/yamnet_256_64x96.h5)  in version 8.1.0 but not in version 8.0.0

`
PS D:\Softwares\en.x-cube-ai-windows_v8.0.0\windows> .\stm32ai.exe export-onnx -m yamnet_256_64x96.h5
Neural Network Tools for STM32AI v1.7.0 (STM.ai v8.0.0-19389)
elapsed time (export-onnx): 1.259s
PS D:\Softwares\en.x-cube-ai-windows_v8.0.0\windows> cd ..\..\en.x-cube-ai-windows_v8.1.0\windows\
PS D:\Softwares\en.x-cube-ai-windows_v8.1.0\windows> .\stm32ai.exe export-onnx -m yamnet_256_64x96.h5
Neural Network Tools for STM32 family v1.7.0 (stm.ai v8.1.0-19520)

INTERNAL ERROR: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
`

aksunlight, 2024-02-26:
The error still occurs with the [new model](https://github.com/STMicroelectronics/stm32ai-modelzoo/blob/main/audio_event_detection/pretrained_models/yamnet/ST_pretrainedmodel_public_dataset/esc10/yamnet_256_64x96_tl/yamnet_256_64x96_tl.h5)
`
PS D:\Softwares\en.x-cube-ai-windows_v8.0.0\windows> .\stm32ai.exe export-onnx -m .\yamnet_256_64x96_tl.h5
Neural Network Tools for STM32AI v1.7.0 (STM.ai v8.0.0-19389)
elapsed time (export-onnx): 1.193s
PS D:\Softwares\en.x-cube-ai-windows_v8.0.0\windows> cd ..\..\en.x-cube-ai-windows_v8.1.0\windows\
PS D:\Softwares\en.x-cube-ai-windows_v8.1.0\windows> .\stm32ai.exe export-onnx -m .\yamnet_256_64x96_tl.h5
Neural Network Tools for STM32 family v1.7.0 (stm.ai v8.1.0-19520)

INTERNAL ERROR: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
`

LMARSTM, 2024-03-04:
Hello, I've been able to reproduce your problem with X-CUBE-AI 8.1.0. The same issue also applies to all .h5 float32 models in the model zoo, so it isn't specific to this particular model.

As this is an issue with X-CUBE-AI and not the model zoo, and the export-onnx command isn't mentioned anywhere in the X-CUBE-AI documentation (though it does show up in the 8.1.0 CLI help),  your issue has been forwarded internally to the appropriate team. I will update this issue when I have more information.

In the meantime, if you want to convert a Keras model from the zoo to ONNX, you can use [tf2onnx](https://github.com/onnx/tensorflow-onnx)

closed 2024-04-09 by LFOSTM