STMicroelectronics/stm32ai-modelzoo #28: Not able to install Tensor flow via CMD

AIStudentbl2, 2024-03-21:
I am trying to use the github for modelzoo using cmd, and I am having trouble with getting the requirements for tensor flow:

[image]

Shahnawax, 2024-03-21:
Hello @AIStudentbl2,
Can you tell us what version of `python` and `pip` you are using and are you connected to internet to get the required packages from pip while running this command? You can get the versions of `python` and `pip` using:
> \> python --version
> \> pip --version

Also, can you try to run pip command with `--no-cache-dir` flag? Like
> \> pip install -r requirements.txt --no-cache-dir

Let us know the answers of these questions so that we can help you better.
Thank you.

AIStudentbl2, 2024-03-21:
Hello @Shahnawax,

Thank you very much for your prompt response I highly appreciate it 😄 

here are the commands you requested:

[image]

Please let me know how to fix these!

Thank you

Shahnawax, 2024-03-22:
@AIStudentbl2 As we indicated in the main `README.md`, that for installing required dependencies without any problem, you have to use `3.9.x ≤ Python versions ≤ 3.10.x` and we recommend to use Python version `3.10.x`. 
Could you install `python 3.10` and give it a try?

AIStudentbl2, 2024-03-25:
Hello @Shahnawax that did the trick, but it also required me to uninstall python 3.12.x and it wouldn't recognise it unless downloaded from the windows store, but after downloading python 3.10 from windows store it started working 👍. Thank you!

closed 2024-03-25 by AIStudentbl2