STMicroelectronics/stm32ai-modelzoo #29: Unable to run Model Zoo onto STM32L562 Board

AIStudentbl2, 2024-03-22:
I have connected my STM32L562 board to my computer to connect to the IDE, and I know there is no issue with the cable as my PC will show that the STM is connected (2nd picture), but the IDE is saying that there is no device connected:
[image]

[image]

LFOSTM, 2024-03-29:
Hello,
This is not a model zoo issue...
Please post your issue here : https://community.st.com/t5/stm32-mcus-machine-learning-ai/bd-p/stm32-mcu-machine-learning-ai-forum
Regards,

closed 2024-03-29 by LFOSTM

LFOSTM, 2024-04-02:
Hello,
For non related ML questions, but related to Cube IDE, , you can post your questions here : https://community.st.com/t5/stm32cubeide-mcus/bd-p/stm32-mcu-cubeide-forum
Thanks,