STMicroelectronics/stm32ai-modelzoo #34: Error: File does not exist: STM32H747I-DISCO_GettingStarted_ObjectDetection_CM7.elf

Darshcg, 2024-05-07:
Hi,

I am trying to run the existing object detection (https://github.com/STMicroelectronics/stm32ai-modelzoo/tree/main/object_detection/deployment) demo with STM32H747I-DISCO and B-CAMS_OMV camera module. 

During flashing the model onto the board, I am getting the below error: 

```
building.. cm7.release
[returned code = 1 - FAILED]
flashing.. cm7.release STM32H747I-DISCO
Board programming failed: "Error: File does not exist: STM32H747I-DISCO_GettingStarted_ObjectDetection_CM7.elf"
```

I followed all the steps mentioned in the readme (https://github.com/STMicroelectronics/stm32ai-modelzoo/tree/main/object_detection/deployment), but not sure what exactly I am missing. Would be great if you could assist me in resolving this issue. 

Thank you!

GRATTINSTM, 2024-05-07:
Hello @Darshcg, 

Which Cube.AI version are you using (tools/stm32ai/version in the YAML file) ? Can you make a try with the version 8.1.0 ? 
If the cloud computation still doesn't work could you install Cube.AI 8.1.0 locally and make a try (tools/stm32ai/on_cloud: False) ? 

Guillaume

Darshcg, 2024-05-07:
Hi @GRATTINSTM, 

Thank you for your response! 

I am using Cube.AI version = 8.1.0, with on_cloud: True.

Yeah sure, I will try installing Cube.AI 8.1.0 locally and set on_cloud: False. 

Thanks!

Darshcg, 2024-05-07:
Hi @GRATTINSTM,

I tried the local installation of Cube.AI 8.1.0 and then setting `on_cloud: False`, then also I am facing the same issue!
```
building.. cm7.release
[returned code = 1 - FAILED]
flashing.. cm7.release STM32H747I-DISCO
Board programming failed: "Error: File does not exist: STM32H747I-DISCO_GettingStarted_ObjectDetection_CM7.elf"

```

GRATTINSTM, 2024-05-07:
Did you follow all the steps of the readme ? Could you sent the user_config.yaml file ?

Darshcg, 2024-05-07:
Hi @GRATTINSTM, Yeah, I followed all the steps of readme(https://github.com/STMicroelectronics/stm32ai-modelzoo/tree/main/object_detection/deployment), below is my `user_config.yaml`:

[user_config.yaml.txt](https://github.com/STMicroelectronics/stm32ai-modelzoo/files/15238040/user_config.yaml.txt)

GRATTINSTM, 2024-05-07:
It seems that many fields are not correctly filled. Please copy/paste the values of each field (except the tools paths) from the readme.

Darshcg, 2024-05-07:
Hi @GRATTINSTM, I put all the same for each field(except the tools paths), but still I am getting the same error. No Sure exactly.

[user_config_updated.yaml.txt](https://github.com/STMicroelectronics/stm32ai-modelzoo/files/15238524/user_config_updated.yaml.txt)

I am also attaching the log below:

```
compiling.. "ssd_mobilenet_v2_fpnlite_035_192_int8_tflite" session
 model_path  : ['../pretrained_models/ssd_mobilenet_v2_fpnlite/ST_pretrainedmodel_public_dataset/coco_2017_person/ssd_mobilenet_v2_fpnlite_035_192/ssd_mobilenet_v2_fpnlite_035_192_int8.tflite']
 tools       : 8.1.0 (x-cube-ai pack)
 target      : "STM32H747I-DISCO Getting Started Object Detection (STM32CubeIDE)" (stm32_cube_ide/cm7.release/stm32h7)
 options     : ../../stm32ai_application_code/object_detection/mempools_STM32H747I-DISCO.json
 results -> RAM=578,812 IO=0:0 WEIGHTS=1,007,872 MACC=122,776,743 RT_RAM=89,604 RT_FLASH=216,626 LATENCY=0.000
[INFO]: Dispatch weights between internal and external flash to fit the large model
[INFO] : Optimized C code + Lib/Inc files generation done.
[INFO] : Building the STM32 c-project..
deploying the c-project.. "STM32H747I-DISCO Getting Started Object Detection (STM32CubeIDE)" (stm32_cube_ide/cm7.release/stm32h7)
updating.. cm7.release
 -> s:copying file.. "network_config.h" to ../../stm32ai_application_code/object_detection/Application/Network/Inc/network_config.h
 -> s:copying file.. "network.h" to ../../stm32ai_application_code/object_detection/Application/Network/Inc/network.h
 -> s:copying file.. "network.c" to ../../stm32ai_application_code/object_detection/Application/Network/Src/network.c
 -> s:copying file.. "network_data.h" to ../../stm32ai_application_code/object_detection/Application/Network/Inc/network_data.h
 -> s:copying file.. "network_data.c" to ../../stm32ai_application_code/object_detection/Application/Network/Src/network_data.c
 -> s:copying file.. "network_data_params.h" to ../../stm32ai_application_code/object_detection/Application/Network/Inc/network_data_params.h
 -> s:copying file.. "network_data_params.c" to ../../stm32ai_application_code/object_detection/Application/Network/Src/network_data_params.c
 -> s:copying file.. "network_generate_report.txt" to ../../stm32ai_application_code/object_detection/Application/Network/Src/network_generate_report.txt
 -> s:removing dir.. ../../stm32ai_application_code/object_detection/Middlewares/ST/STM32_AI_Runtime/Lib
 -> s:copying dir.. "Lib" to ../../stm32ai_application_code/object_detection/Middlewares/ST/STM32_AI_Runtime/Lib
 -> s:removing dir.. ../../stm32ai_application_code/object_detection/Middlewares/ST/STM32_AI_Runtime/Inc
 -> s:copying dir.. "Inc" to ../../stm32ai_application_code/object_detection/Middlewares/ST/STM32_AI_Runtime/Inc
 -> u:copying file.. "ai_model_config.h" to ../../stm32ai_application_code/object_detection/Application/STM32H747I-DISCO/Inc/CM7/ai_model_config.h
 -> updating cproject file "/home/darshan/stm32ai-modelzoo/stm32ai_application_code/object_detection/Application/STM32H747I-DISCO/STM32CubeIDE/CM7" with "NetworkRuntime810_CM7_GCC.a"
building.. cm7.release
[returned code = 1 - FAILED]
flashing.. cm7.release STM32H747I-DISCO
Board programming failed: "Error: File does not exist: STM32H747I-DISCO_GettingStarted_ObjectDetection_CM7.elf"

```

Darshcg, 2024-05-07:
Just to mention that: I tried image classification application: and it generates: `STM32H747I-DISCO_GettingStarted_ImageClassification_CM7.elf` but not getting generated for Object detection application

Darshcg, 2024-05-08:
Hi @GRATTINSTM,

Seems like the issue was with Linux and STM IDE and Cube.AI compatibility. I did the setup in windows with IDE version = 1.14.0 and Cube AI version = 8.1.0 and it works fine!

Thank you!

GRATTINSTM, 2024-05-13:
Hi @Darshcg, 

I'm glad it works. We will investigate the reason why it is not working on your Linux setup. 

Guillaume

Darshcg, 2024-05-22:
Hi @GRATTINSTM,

Thank you for your response.

I actually followed the links provided in the Readme for the software requirements:
https://github.com/STMicroelectronics/stm32ai-modelzoo/blob/main/object_detection/deployment/README.md

[image]

closed 2024-10-03 by GRATTINSTM