STMicroelectronics/stm32ai-modelzoo #37: B cam omv + USB Display

Vietsol-AnhHaiLe, 2024-06-12:
hello,

I have issue when i ran example image classsification for stm32h743zi2

my error is 
[image]
 can you help me, tks alot

Vietsol-AnhHaiLe, 2024-06-17:
> Hello,
> Could you provide more information of what you did?
> Thanks

Hello,

Im trying stm32-Model zoo examples but i have some problems

Image classification - FdMobilenet : I have generated code from image_classification.ipynb and flash the program success

But after i flash, my hardware seem get in to loop that can't find camera init, i have checked wires for many times and it works well, but my board NUCLEO-H743ZI2  + B-Cam OMV + USB Display stop at welcome screen

Can you help me!

Thanks

Vietsol-AnhHaiLe, 2024-06-17:
> Hello @Vietsol-AnhHaiLe<https://github.com/Vietsol-AnhHaiLe>,
>
> I just tried to deploy thanks to the Jupyter Notebook image_classification.ipynb and everything works well. Please take care to use cables as short as possible to avoid EMI.
>
> Guillaume

can you record a short video about your live project please

Vietsol-AnhHaiLe, 2024-06-17:
my generate success but it doesn't work
[cid:d309dc56-2b06-40f1-8bf3-062626153774]

[cid:426c92d4-e50f-4294-a53d-a601aae78181]

Vietsol-AnhHaiLe, 2024-06-17:
[image]
[image]

Vietsol-AnhHaiLe, 2024-06-17:
[image]

GRATTINSTM, 2024-06-18:
Please find attached the project I tested on my NUCLEO-H743ZI2 and working as expected. 
[image_classification.zip](https://github.com/user-attachments/files/15886526/image_classification.zip)

Guillaume

Vietsol-AnhHaiLe, 2024-06-28:
Hello @***@***.***>, Im trying STM32F746G_Discovery + B-CAM-OMVS => Image classification

Im a student and im interested in Image CLassification on Hardware and im trying to using it á an sample for my final Uni projects but i dont have much experience on Hardware, coud your team please doing a samples project with my hardware, tks a lot

closed 2024-09-25 by GRATTINSTM