STMicroelectronics/stm32ai-modelzoo #44: Tensorflow version not found

TECO-Octo, 2024-07-19:
Hello, I am on MacOS (python 3.10.8) and when I want to install the `requirements.txt`, I get this error :

```
ERROR: Could not find a version that satisfies the requirement tensorflow==2.8.3 (from versions: 2.13.0rc0, 2.13.0rc1, 2.13.0rc2, 2.13.0, 2.13.1, 2.14.0rc0, 2.14.0rc1, 2.14.0, 2.14.1, 2.15.0rc0, 2.15.0rc1, 2.15.0, 2.15.1, 2.16.0rc0, 2.16.1, 2.16.2, 2.17.0rc0, 2.17.0rc1, 2.17.0)
ERROR: No matching distribution found for tensorflow==2.8.3

```
If I update the version on tensorflow in the `requirements.txt`, I assume there will be conflicts with the other packages version (most likely newer version needed)

Any idea?

PS : I will try to update all the packages in order to make them work with a newer version of tensorflow, hoping that the code will work after that

LFOSTM, 2024-09-18:
Hello,
Could you start from a clean environment following installation recommendations from the main readme:
git clone https://github.com/STMicroelectronics/stm32ai-modelzoo.git
cd stm32ai-modelzoo
python -m venv st_zoo
source st_zoo/bin/activate
pip install -r requirements.txt

Regards

LFOSTM, 2024-10-15:
Hello,
Can you confirm that the issue is solved on your side? Then I will close the ticket.
Regards,
Laurent

bayesiandog, 2024-11-06:
I previously had the same issue and solved it by downgrading python to 3.10. Make sure you then create an environment that runs that version. pip python --version will show you which version is currently active.

closed 2025-06-02 by Shahnawax