STMicroelectronics/stm32ai-modelzoo #46: Hardfault in ILI9341_STM32_Driver.c

clementfumey, 2024-09-23:
In function ILI9341_Draw_Colour_Burst initialization at 0 of Buffer_Size, cause division by 0 in Sending_in_Block
	
```
	uint32_t Buffer_Size = 0;
	unsigned char chifted = 	Colour>>8;
	unsigned char burst_buffer[Buffer_Size];
	uint32_t Sending_Size = Size*2;
	uint32_t Sending_in_Block = Sending_Size/Buffer_Size;
```

```
if((Size*2) < BURST_MAX_SIZE)
{
	Buffer_Size = Size;
}
else
{
	Buffer_Size = BURST_MAX_SIZE;
}
```
Should be placed before

GRATTINSTM, 2024-09-24:
Hello @clementfumey, 

As stated in the header of ILI9341_STM32_Driver.c, this code has been first developed by [Matej Artnak in 2017](https://github.com/martnak/STM32-ILI9341/tree/master/Src/ILI9341) and is provided as is. 
ST will not provide any support for the issues you may have using this drivers. 

Guillaume

closed 2024-09-24 by GRATTINSTM

GRATTINSTM, 2024-09-24:
We may have introduced some bugs in the code as we didn't test all the functions after the modifications we have made. Our mistake has been to let all the functions available, even the ones that are not used in the ModelZoo. It will be corrected in the next release. Thank you for your feedback. 

Guillaume