STMicroelectronics/stm32ai-modelzoo #52: Error in development Object Detection using the given example

Hood7319, 2024-11-11:
Hello,

I want to deploy a pre-trained object detection model on an STM32 board using STM32Cube.AI.

I deploy the model [ssd_mobilenet_v2_fpnlite_035_192_int8.tflite] pretrained on COCO dataset using the necessary parameters provided in [ssd_mobilenet_v2_fpnlite_035_192_config.yaml] as the example, but I got the following errors:

I only change the deployment_config.yaml.

[deployment_config.yaml.txt](https://github.com/user-attachments/files/17702947/deployment_config.yaml.txt)

[INFO] : Successfully connected!
[INFO] : Starting the model memory footprints estimation...
Error executing job with overrides: []
Traceback (most recent call last):
  File "E:\Desktop\stm32ai-modelzoo-main\object_detection\src\stm32ai_main.py", line 234, in main
    process_mode(cfg,
  File "E:\Desktop\stm32ai-modelzoo-main\object_detection\src\stm32ai_main.py", line 113, in process_mode
    deploy(cfg)
  File "E:\Desktop\stm32ai-modelzoo-main\object_detection\src\../deployment\deploy.py", line 98, in deploy
    stm32ai_deploy(target=board, stlink_serial_number=stlink_serial_number, stm32ai_version=stm32ai_version, c_project_path=c_project_path,
  File "E:\Desktop\stm32ai-modelzoo-main\object_detection\src\../../common/deployment\common_deploy.py", line 378, in stm32ai_deploy
    dispatch_weights(internalFlashSizeFlash_KB=board.config.internalFlash_size,
  File "E:\Desktop\stm32ai-modelzoo-main\object_detection\src\../../common/deployment\common_deploy.py", line 58, in dispatch_weights
    sorted_weights = dict(sorted(graph["weights"].items(), key=lambda item: item[1]['buffer_c_count'], reverse=True))
  File "E:\Desktop\stm32ai-modelzoo-main\object_detection\src\../../common/deployment\common_deploy.py", line 58, in <lambda>
    sorted_weights = dict(sorted(graph["weights"].items(), key=lambda item: item[1]['buffer_c_count'], reverse=True))
KeyError: 'buffer_c_count'

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

How can I sovle the problem? 

Thank you for your guidance.

Vietsol-AnhHaiLe, 2024-12-16:
I have the same problem

h-arthi, 2025-01-23:
Hey I'm having the same issue, anyone have any luck on figuring it out?

LFOSTM, 2025-01-23:
Hello,
Could you make a try using the latest version of the model zoo, and pointing on stedgeai version 10.0.0, please?
Thanks & Regards,

woshuhua, 2025-03-22:
> Hello, Could you make a try using the latest version of the model zoo, and pointing on stedgeai version 10.0.0, please? Thanks & Regards,

I am using the latest version of the model zoo and pointing on st edge AI version 10.0.0. I am getting the same error too...
here's the error: 
[INFO] : Successfully connected!
[INFO] : Starting the model memory footprints estimation...
Error executing job with overrides: []
Traceback (most recent call last):
  File "C:\Users\xuhua\AppData\Local\Programs\Python\Python310\lib\site-packages\clearml\binding\hydra_bind.py", line 230, in _patched_task_function
[INFO] : Starting the model memory footprints estimation...
Error executing job with overrides: []
Traceback (most recent call last):
  File "C:\Users\xuhua\AppData\Local\Programs\Python\Python310\lib\site-packages\clearml\binding\hydra_bind.py", line 230, in _patched_task_function
Error executing job with overrides: []
Traceback (most recent call last):
  File "C:\Users\xuhua\AppData\Local\Programs\Python\Python310\lib\site-packages\clearml\binding\hydra_bind.py", line 230, in _patched_task_function
Traceback (most recent call last):
  File "C:\Users\xuhua\AppData\Local\Programs\Python\Python310\lib\site-packages\clearml\binding\hydra_bind.py", line 230, in _patched_task_function
    return task_function(a_config, *a_args, **a_kwargs)
  File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src\stm32ai_main.py", line 228, in main
    return task_function(a_config, *a_args, **a_kwargs)
  File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src\stm32ai_main.py", line 228, in main
  File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src\stm32ai_main.py", line 228, in main
    process_mode(cfg)
    process_mode(cfg)
  File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src\stm32ai_main.py", line 102, in process_mode
    deploy(cfg)
  File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src\../deployment\deploy.py", line 101, in deploy
    stm32ai_deploy(target=board, stlink_serial_number=stlink_serial_number, stm32ai_version=stm32ai_version, c_project_path=c_project_path,
  File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src\../../common/deployment\common_deploy.py", line 378, in stm32ai_deploy
    dispatch_weights(internalFlashSizeFlash_KB=board.config.internalFlash_size,
  File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src\../../common/deployment\common_deploy.py", line 58, in dispatch_weights
    sorted_weights = dict(sorted(graph["weights"].items(), key=lambda item: item[1]['buffer_c_count'], reverse=True))
  File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src\../../common/deployment\common_deploy.py", line 58, in <lambda>
    sorted_weights = dict(sorted(graph["weights"].items(), key=lambda item: item[1]['buffer_c_count'], reverse=True))
KeyError: 'buffer_c_count'

[deployment_config.yaml.txt](https://github.com/user-attachments/files/19403493/deployment_config.yaml.txt)

woshuhua, 2025-03-23:
> > Hello, Could you make a try using the latest version of the model zoo, and pointing on stedgeai version 10.0.0, please? Thanks & Regards,
> 
> I am using the latest version of the model zoo and pointing on st edge AI version 10.0.0. I am getting the same error too... here's the error: [INFO] : Successfully connected! [INFO] : Starting the model memory footprints estimation... Error executing job with overrides: [] Traceback (most recent call last): File "C:\Users\xuhua\AppData\Local\Programs\Python\Python310\lib\site-packages\clearml\binding\hydra_bind.py", line 230, in _patched_task_function [INFO] : Starting the model memory footprints estimation... Error executing job with overrides: [] Traceback (most recent call last): File "C:\Users\xuhua\AppData\Local\Programs\Python\Python310\lib\site-packages\clearml\binding\hydra_bind.py", line 230, in _patched_task_function Error executing job with overrides: [] Traceback (most recent call last): File "C:\Users\xuhua\AppData\Local\Programs\Python\Python310\lib\site-packages\clearml\binding\hydra_bind.py", line 230, in _patched_task_function Traceback (most recent call last): File "C:\Users\xuhua\AppData\Local\Programs\Python\Python310\lib\site-packages\clearml\binding\hydra_bind.py", line 230, in _patched_task_function return task_function(a_config, *a_args, **a_kwargs) File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src\stm32ai_main.py", line 228, in main return task_function(a_config, *a_args, **a_kwargs) File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src\stm32ai_main.py", line 228, in main File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src\stm32ai_main.py", line 228, in main process_mode(cfg) process_mode(cfg) File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src\stm32ai_main.py", line 102, in process_mode deploy(cfg) File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src../deployment\deploy.py", line 101, in deploy stm32ai_deploy(target=board, stlink_serial_number=stlink_serial_number, stm32ai_version=stm32ai_version, c_project_path=c_project_path, File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src../../common/deployment\common_deploy.py", line 378, in stm32ai_deploy dispatch_weights(internalFlashSizeFlash_KB=board.config.internalFlash_size, File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src../../common/deployment\common_deploy.py", line 58, in dispatch_weights sorted_weights = dict(sorted(graph["weights"].items(), key=lambda item: item[1]['buffer_c_count'], reverse=True)) File "C:\Users\xuhua\Documents\stm32ai-modelzoo-services\object_detection\src../../common/deployment\common_deploy.py", line 58, in sorted_weights = dict(sorted(graph["weights"].items(), key=lambda item: item[1]['buffer_c_count'], reverse=True)) KeyError: 'buffer_c_count'
> 
> [deployment_config.yaml.txt](https://github.com/user-attachments/files/19403493/deployment_config.yaml.txt)

This issue is disappeared when I change the model to st_ssd_mobilenet_v1_025_192_int8.tflite. 
I am not sure about the model extraction have any problem that caused the error buffer_c_count.

GHORTESSTM, 2025-03-25:
Hello,

The issue has been solved in the last version: https://github.com/STMicroelectronics/stm32ai-modelzoo-services.
You fetch the last version.

The problem was present when using the dev cloud. Now it works both with stedgeai in local and with the dev cloud.