STMicroelectronics/stm32ai-modelzoo #53: [INFO] Bypassing Edge AI Cloud  - To create an account, go to https://stedgeai-dc.st.com/home. Enter your credentials:

Raulpb49, 2025-01-16:
How to use stedgeai on PC instead of stedgeaicloud? I see this is coming somewhere form the \stm32ai-modelzoo-services\common\stm32ai_dc\backend\cloud . Is it possible to by pass the cloud logging and use the local edgeai?

I dont want to use the ST Edge AI Cloud.

Best,
R

LFOSTM, 2025-01-16:
Hello,
You can deploy your model locally specifying on_cloud : False in the yaml file.
May I ask why you don't want to use the ST Edge AI Cloud?
Regards,
Laurent

Raulpb49, 2025-01-16:
Thanks for your prompt response.
I am facing some https connection issue while  connecting to it . I am
behind the proxy but i have already set the proxy (http,https)and also the
username and password to my st.com in the environment settings ( I am on
windows) . It fails after 5 retries.
Regards,

Shahnawax, 2025-01-17:
@Raulpb49 can you try few things. 
- add `ssl_verify:false` and `no_ssl_verify:1` in your environment variables.
- use the http://username:password@proxy.com:port string for both http and https proxies
- make sure your credentials for `stmai_username` and `stmai_password` are upto date by logging in to the webpage through a web browser [https://stedgeai-dc.st.com/home](https://stedgeai-dc.st.com/home) 
- Finally, log off and log on to make sure that the latest environment variable changes are taken into account.

- Alternatively, you can check with your IT to understand what is the issue with proxy verification and use the local installation in the meanwhile by just providing the correct path in the variable path_to_stedgeai and setting the on_cloud variable to False.

__NOTE__: You do not need to encode the passwords for the myST account but you need to encode the password for your proxy account in your proxy strings. For example a password 123&st has to be written as 123%26st in the proxy string.

I hope this solves the issue.

Raulpb49, 2025-01-20:
>    use the ***@***.***:port string for both http

Thanks Muhammad ,
I will check this out. So Far i am using the desktop version so I think it
shouldn't be a problem right? Both the cloud and desktop version are the
same in functionalities - correct?
If that is so I will stick with the desktop version as of now.
Regards,
R

Shahnawax, 2025-01-20:
That is true, functionality wise there is no difference. The benefit of using the dev cloud is that you can benchmark on different boards without actually having the boards in hand. However, for generating the code, and doing the deployment, local vs dev cloud will provide the same outcome.

Raulpb49, 2025-01-21:
Thanks @Shahnawax  
Understood. Anyway, the cloud one I can use from the web which I think is working fine. I will use it to benchmark on different platform.
I also have one question related to ST Edge AI Core installation: While installing the ST Edge AI core on windows, I have to manually browse the **en.n6-ai-getstarted-v1.0.0.zip** package on windows. 
Now I am on headless Ubuntu setup with no access to GUI, is it possible to install the edge AI core on Ubuntu by providing this .zip package as option to browse or I need to have a GUI interface enabled to select it and proceed for installation mandatorily?
Regards,

Shahnawax, 2025-01-21:
@Raulpb49  you are welcome.

The installer supports headless mode also. Download the online installer for linux and the zip for STM32N6***neuralart extract the contents and then try something like:

`./stedgeai-linux-onlineinstaller install stedgeai0200.stneuralart`

You can find the documentation here:

[https://stedgeai-dc.st.com/assets/embedded-docs/modular_installer.html](https://stedgeai-dc.st.com/assets/embedded-docs/modular_installer.html)

Raulpb49, 2025-01-21:
Hello @Shahnawax  , 
./stedgeai-linux-onlineinstaller install stedgeai0200.stneuralart gives below error.
./stedgeai-linux-onlineinstaller: error while loading shared libraries: libxcb-image.so.0: cannot open shared object file: No such file or directory
There was errors related to other shared libs(libxkbcommon-x11.so.0), i tried to resolve them but after fixing some , i get above.
The documentation says: 
mkdir -p $CONDA_PREFIX/etc/conda/activate.d
echo 'export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$CONDA_PREFIX/lib/' > $CONDA_PREFIX/etc/conda/activate.d/env_vars.sh 

I did this as well but isn't these are for cuda? i am on x64 Intel PC.

Regards,

Shahnawax, 2025-02-13:
Great! Enjoy using model zoo and let us know if there are any other issues. 
Closing issue as solved...
Thank you.