STMicroelectronics/stm32ai-modelzoo #55: Cannot reproduce emnist model.

SheperdHannigan, 2025-03-18:
import scipy.io
import numpy as np
import os
import tensorflow as tf
from tensorflow.keras.optimizers import Adam

def load_emnist_subset(mat_file_path):
    """
    Loads the EMNIST ByClass dataset and extracts a subset containing only digits [0-9] and capital letters [A-Z].
    Returns both original and normalized images along with labels in both normal and one-hot formats.

    Parameters:
        mat_file_path (str): Path to the 'emnist-byclass.mat' file.

    Returns:
        (X_train, X_test, X_train_normalized, X_test_normalized, y_train, y_test, y_train_onehot, y_test_onehot):
            - X_train: (N_train, 28, 28), original images in range [0, 255].
            - X_test: (N_test, 28, 28), original images in range [0, 255].
            - X_train_normalized: (N_train, 28, 28), normalized images in range [-1, 1].
            - X_test_normalized: (N_test, 28, 28), normalized images in range [-1, 1].
            - y_train: (N_train,), original labels.
            - y_test: (N_test,), original labels.
            - y_train_onehot: (N_train, 36), one-hot encoded labels.
            - y_test_onehot: (N_test, 36), one-hot encoded labels.
    """
    # 1. Load the MATLAB .mat file
    mat_data = scipy.io.loadmat(mat_file_path)

    # 2. Access the 'dataset' structure
    dataset = mat_data['dataset']

    # Load training set
    X_train = dataset[0, 0]['train'][0, 0]['images']
    y_train = dataset[0, 0]['train'][0, 0]['labels'].flatten()

    # Load test set
    X_test = dataset[0, 0]['test'][0, 0]['images']
    y_test = dataset[0, 0]['test'][0, 0]['labels'].flatten()

    # 3. Select only [0-9] and [A-Z] classes
    valid_classes = np.concatenate([np.arange(0, 10), np.arange(10, 36)])  # Digits [0-9] + Capital letters [A-Z]

    # Filter training set
    train_mask = np.isin(y_train, valid_classes)
    X_train = X_train[train_mask]
    y_train = y_train[train_mask]

    # Filter test set
    test_mask = np.isin(y_test, valid_classes)
    X_test = X_test[test_mask]
    y_test = y_test[test_mask]

    # 4. Reshape images to (N, 28, 28)
    X_train = X_train.reshape(-1, 28, 28)
    X_test = X_test.reshape(-1, 28, 28)

    # 5. Copy original dataset (pixel values in range [0, 255])
    X_train_original = X_train.copy()
    X_test_original = X_test.copy()

    # 6. Apply rescaling: scale=1/127.5, offset=-1
    X_train_normalized = X_train.astype(np.float32) * (1 / 127.5) - 1
    X_test_normalized = X_test.astype(np.float32) * (1 / 127.5) - 1

    # 7. Remap labels to [0, 35] to ensure one-hot encoding continuity
    label_map = {label: idx for idx, label in enumerate(valid_classes)}
    y_train_mapped = np.array([label_map[y] for y in y_train])
    y_test_mapped = np.array([label_map[y] for y in y_test])

    # 8. Convert to one-hot encoding
    num_classes = len(valid_classes)  # Should be 36
    y_train_onehot = np.eye(num_classes)[y_train_mapped]  # (N_train, 36)
    y_test_onehot = np.eye(num_classes)[y_test_mapped]  # (N_test, 36)

    return X_train_original, X_test_original, X_train_normalized, X_test_normalized, y_train, y_test, y_train_onehot, y_test_onehot

# Example usage:
mat_file = os.path.expanduser(emnist-byclass.mat)
X_train, X_test, X_train_norm, X_test_norm, y_train, y_test, y_train_onehot, y_test_onehot = load_emnist_subset(mat_file)

# Output dataset shapes and value range checks
print(f"Train dataset shape: {X_train.shape}, Test dataset shape: {X_test.shape}")
print(f"Train normalized shape: {X_train_norm.shape}, Test normalized shape: {X_test_norm.shape}")
print(f"Train labels shape: {y_train.shape}, One-hot shape: {y_train_onehot.shape}")
print(f"Test labels shape: {y_test.shape}, One-hot shape: {y_test_onehot.shape}")
print(f"Unique classes in train: {np.unique(y_train)}")
print(f"Unique classes in test: {np.unique(y_test)}")
print(f"Data range (original train): min={X_train.min()}, max={X_train.max()}")  # Should be 0 - 255
print(f"Data range (normalized train): min={X_train_norm.min()}, max={X_train_norm.max()}")  # Should be -1 to 1

def keras_evaluate(model_path, input_data, input_label):
    model = tf.keras.models.load_model(model_path)
    optimizer = Adam(learning_rate = 0.01)
    model.compile(optimizer=optimizer,
                  loss='categorical_crossentropy',
                  metrics=['accuracy'])
    loss, accuracy = model.evaluate(input_data, input_label)
    return loss, accuracy

model_save_path = os.path.expanduser(MODEL_PATH)
# x_train, x_test, x_train_normalized, x_test_normalized, y_train, y_test, y_train_onehot, y_test_onehot = load_dataset("emnist_byclass")

loss, accuracy = keras_evaluate(model_save_path + "st_mnist_v1_28_tfs.h5", X_train_norm, y_train_onehot)

output:
16688/16688 [==============================] - 24s 1ms/step - loss: 11.9799 - accuracy: 0.1579

LFOSTM, 2025-03-18:
Hello,
Can you clarify the Pb or error, please?
Thanks

SheperdHannigan, 2025-03-18:
Yes, as you can see, the accuracy is 0.1579. But according to the [README.md](https://github.com/STMicroelectronics/stm32ai-modelzoo/tree/main/image_classification/st_mnist), the accuracy should be about 91.89%.

LFOSTM, 2025-03-18:
OK clear.
Can you share the yaml file you used in your experiment?
Thanks

SheperdHannigan, 2025-03-18:
Hi, @LFOSTM , I did not use the yaml file in my experiments. I load the model from image_classification/st_mnist/ST_pretrainedmodel_public_dataset/emnist_byclass/st_mnist_v1_28_tfs/st_mnist_v1_28_tfs.h5. At the same time I load the emnist_byclass.mat dataset using scipy from https://www.nist.gov/itl/products-and-services/emnist-dataset. 

I strictly follow the yaml file to apply   rescaling: { scale: 1/127.5, offset: -1 } to the original image data and I got normalized image data ranage from -1.0 to 1.0. And then I feed the normalized image data into the pretrained .h5 model provided by the stm32. However, I got extremly low accuracy.

I do this process using other image_classification models provided stm32, and the other models run well. Only the emnsit_byclass model has extremely low accuracy.

SheperdHannigan, 2025-03-18:
Do you mean the [stm32ai-modelzoo-services](https://github.com/STMicroelectronics/stm32ai-modelzoo-services) repo? 
However, I cannot find a exsisted evaluation .yaml for  st_emnist_byclass model.

Moreover, I wonder if the evaluation service is different from the code I demonstrated above.

LFOSTM, 2025-03-19:
Yes exactly - this repo is the one to use and benefit from all the model zoo services like the training, evaluation, prediction, quantization, benchmark (localy or using the developper clout) and deployment.
I recommend to use a config file example for evaluation like this one : https://github.com/STMicroelectronics/stm32ai-modelzoo-services/blob/main/image_classification/src/config_file_examples/evaluation_config.yaml
And modify it for your needs like your model_path, ... as precised in the readme.md (https://github.com/STMicroelectronics/stm32ai-modelzoo-services/blob/main/image_classification/src/README.md): "The dataset_name attribute is optional unless you want to use the EMNIST, CIFAR-10, and CIFAR-100 datasets provided with the Model Zoo. In this case, you need to set it to either 'emnist', 'cifar10', or 'cifar100'.
You can be inspired from the .yaml used for the training of st_mnist on emnist dataset (https://github.com/STMicroelectronics/stm32ai-modelzoo/blob/main/image_classification/st_mnist/ST_pretrainedmodel_public_dataset/emnist_byclass/st_mnist_v1_28_tfs/st_mnist_v1_28_tfs_config.yaml) to get some fields like the class_names and the preprocessing. 

Regards,
Laurent

SheperdHannigan, 2025-03-19:
Hi @LFOSTM,

Thank you for your kind explanation. I carefully revisited the .yaml file and specifically the corresponding emnist data loading function in data_loader.py. I found that the numpy reshape option with order='F' was preventing me from reproducing the accuracy. After adding order='F' to the reshape method, I was able to achieve the correct accuracy.

Thanks!
Sheperd Hannigan

closed 2025-03-19 by SheperdHannigan